pyavitz/rpi-img-builder #2: make-scripts breaks on more recent kernels

AlwaysLivid, 2020-04-16:
So, as far as I am aware of, the way `make-scripts` works is by applying certain patches to the header sources before compiling them.

This seems to result in unexpected behavior in more recent kernel versions (e.g. `Linux 5.6.2`)

```
~$ sudo make-scripts

Patching and running make scripts on the kernel headers.
The next patch would create the file tools/include/tools/be_byteshift.h,
which already exists!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored
The next patch would create the file tools/include/tools/le_byteshift.h,
which already exists!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.host:124: scripts/kconfig/conf.o] Error 2
make[2]: *** Deleting file 'scripts/kconfig/conf.o'
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.host:124: scripts/kconfig/expr.o] Error 2
make[2]: *** Deleting file 'scripts/kconfig/expr.o'
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.host:124: scripts/kconfig/confdata.o] Error 2
make[2]: *** Deleting file 'scripts/kconfig/confdata.o'
make[1]: *** [Makefile:568: syncconfig] Error 2
make: *** [Makefile:678: include/config/auto.conf.cmd] Error 2
Done.
```

pyavitz, 2020-04-16:
Interesting. It seems as though the patch has already been applied and no longer needed in 5.6.y. Although it doesn't seem like running make scripts actually works? So something isn't right.

I know on mainline compiling against the headers is also broken at the moment.

AlwaysLivid, 2020-04-16:
> Interesting. It seems as though the patch has already been applied and no longer needed in 5.6.y. Although it doesn't seem like running make scripts actually works? So something isn't right.
> 
> I know on mainline compiling against the headers is also broken at the moment.

Yup, I discovered this after dealing with difficulties with the `wireguard-dkms` package.

pyavitz, 2020-04-16:
I'm assuming ur using a Pi4? Try this [kernel](http://www.mediafire.com/file/yn7tdmzptiuk58y/rpi4b-linux-image-5.6.2-v8.tar.xz/file). It isn't cross-compiled, it's native.

```
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.6.2-v8 KSRC=/lib/modules/5.6.2-v8/build....................................................
cleaning build area...

DKMS: build completed.

88XXau.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.2-v8/updates/dkms/

depmod...

DKMS: install completed.
Finished running dkms install steps.
```

AlwaysLivid, 2020-04-16:
> I'm assuming ur using a Pi4? Try this [kernel](http://www.mediafire.com/file/yn7tdmzptiuk58y/rpi4b-linux-image-5.6.2-v8.tar.xz/file). It isn't cross-compiled, it's native.

Should a script be provided additionally for the time being?

pyavitz, 2020-04-16:
A script to compile native? You can already do that with the current kernel script. Would just need to compile it on the Pi4 or another SBC running aarch64.

You can actually run the whole build script from the Pi4 if you like. Grab the aarch64 IMG from the hub, git clone and ur off to the races.

pyavitz, 2020-04-20:
In my experienced this issue comes and goes through out kernel development and being as such, I'm going to close this.

pyavitz, 2020-04-30:
Fixed in `edge branch`
```
  MODPOST 1 modules
  CC [M]  /home/patrick/rtl8812au/88XXau.mod.o
  LD [M]  /home/patrick/rtl8812au/88XXau.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.6.7-v8'
```

closed 2020-04-30 by pyavitz