pyavitz/rpi-img-builder #3: Provide option to compile kernels into .deb files

AlwaysLivid, 2020-04-18:


AlwaysLivid, 2020-04-18:
I'd like to work on this, if that's not too big of a deal.

My approach would be the following;
* create new option in Makefile that enables an environment variable and then runs the `make-kernel` script.
* if the said environment variable has been toggled, run the following command on the root of the kernel sources; `fakeroot debian/rules binary`

pyavitz, 2020-04-18:
Sounds good.

If ur going to be working directly from the hub, please use feature branch.

**Branches**:
master - default
feature
rpi4b - depreciated 
rpi3b - depreciated
rpi0w - depreciated

AlwaysLivid, 2020-04-18:
> Sounds good.
> 
> If ur going to be working directly from the hub, please use feature branch.
> 

I'll open a pull request, so that you'll be able to tell apart tiny corrections and details before it gets merged, since I'm still a bit unfamiliar with the codebase.

pyavitz, 2020-04-26:
**Mission accomplished!**

Feature branch is now ready for testing.

closed 2020-04-28 by pyavitz