pyavitz/rpi-img-builder #5: stage 1 throws error with distribution "bullseye"

alex9434, 2020-04-20:
I am trying to install the testing distribution "bullseye". I have already commented out the repositories that are not available in testing. However, towards the end I get the following error:

```
Installing firmware and needed packages.
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package btrfs-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'btrfs-tools' has no installation candidate
make: *** [Makefile:116: image] Error 100
```
Any idea why we need to install 'btrfs-tools'? How to best solve this error?

pyavitz, 2020-04-20:
That is because the package name has been changed: [btrfs-progs](https://packages.debian.org/bullseye/btrfs-progs)

To fix it, either remove `btrfs-tools` or rename it to `btrfs-progs`.
You will find it on line 151 of `scripts/rpi4-stage2`.

When dealing with debian unstable ur bound to run into issues here and there concerning package name changes and other slight diffs.  I can't promise things will build correctly or run as they should in this use case.

closed 2020-04-20 by pyavitz

alex9434, 2020-04-22:
That worked. Thank you!