pyavitz/rpi-img-builder #7: energy consumption

alex9434, 2020-04-22:
One more quick question: I get a lot of undervolt warnings (yellow lightning bolt) with this image. I never got a warning with the original raspbian image. Besides switching power supply, do you have any recommendations which of your changes compared to raspbian cased the increased power demand so that I can change them back?

pyavitz, 2020-04-22:
Interesting.

I myself have had the opposite experience since switching to Debian. With the exception of one of my Pi3Bs which has always "spit lightning" no matter the OS.

I honestly change very lil and as far as I know, nothing that would effect power consumption. In all honesty, I try to keep it as much like the default Raspbian setup as I can on the config side of things, whilst still trying to maintain that vanilla Debian feel.

Even the current defconfig used in the `master branch` is basically the default raspbian defconfig with initrd removed, as it doesn't really play nice with FAT partitions. With that said, I decided to recently change this in the `feature branch` and to give people the option to choose between initrd and no initrd or a custom_defconfig.

Have you tried building or using the armv7l/armhf img? Do you get the same results?

First and foremost though, I would try switching out the [PSU](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md) and to always use one that is up to snuff. No phone chargers!

alex9434, 2020-05-04:
Closing this issue. Swapping out the power supply solved my problem. Still curious why this did not pop up with the original Raspbian distribution.

closed 2020-05-04 by alex9434