pyavitz/rpi-img-builder #9: Missing hardware decoding in Chromium

alex9434, 2020-04-29:
Hi, I noticed that the hardware support for video decoding in Chromium is missing in your build. Here is what I get on chrome://gpu

```
1. Raspbian
Canvas: Hardware accelerated
Flash: Hardware accelerated
Flash Stage3D: Hardware accelerated
Flash Stage3D Baseline profile: Hardware accelerated
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
Out-of-process Rasterization: Disabled
Hardware Protected Video Decode: Unavailable
Rasterization: Software only. Hardware acceleration disabled
Skia Renderer: Disabled
Video Decode: Unavailable
Viz Display Compositor: Enabled
Viz Hit-test Surface Layer: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
```

```
2. rpi-img-builder
Canvas: Software only, hardware acceleration unavailable
Flash: Unavailable
Flash Stage3D: Unavailable
Flash Stage3D Baseline profile: Unavailable
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Enabled
Out-of-process Rasterization: Unavailable
OpenGL: Unavailable
Hardware Protected Video Decode: Unavailable
Rasterization: Hardware accelerated
Skia Renderer: Enabled
Video Decode: Unavailable
Viz Display Compositor: Enabled
Vulkan: Disabled
WebGL: Hardware accelerated but at reduced performance
WebGL2: Unavailable
```
What can be done to enable full hardware support in rpi-img-builder

pyavitz, 2020-04-29:
This is beyond the scope of this project and seems like an issue to be taken up with the chromium maintainer over at debian.

Are you trying to watch youtube videos? I've heard people have had success using [Vivaldi](https://vivaldi.com/blog/snapshots/vivaldi-2-5-release-candidate/)'s arm64 release.

Beyond that, you would need to build chromium ur self, which I imagine would be a lot of work.

pyavitz, 2020-04-29:
Not sure if this would work, but raspberrypi provides the [source](http://archive.raspbian.org/raspbian/pool/main/c/chromium-browser/) for the build of chromium they use.

These are the two archives of interest:
[chromium-browser_71.0.3578.80-1~deb9u1.debian.tar.xz](http://archive.raspbian.org/raspbian/pool/main/c/chromium-browser/chromium-browser_71.0.3578.80-1~deb9u1.debian.tar.xz)
[chromium-browser_71.0.3578.80.orig.tar.xz](http://archive.raspbian.org/raspbian/pool/main/c/chromium-browser/chromium-browser_71.0.3578.80.orig.tar.xz)

The debian tarball is the one that contains all the patches.

alex9434, 2020-04-29:
OK, thank you very much for these tipps!

closed 2020-04-29 by alex9434