pyavitz/rpi-img-builder #10: Sound and Bluetooth not working

alex9434, 2020-05-06:
Hi, first of all thank you for this great project! Apologies for another question: I cannot get sound and bluetooth to work. 
1. Sound: I have installed the full mate desktop (apt-get install task-mate-desktop). AlsaMixer does recognize bcm2835 sound card. However, there is no sound on the headphone jack:
```
┌────────────────────────────── AlsaMixer v1.2.2 ──────────────────────────────┐
│ Card: PulseAudio                                     F1:  Help               │
│ Chip: PulseAudio                                     F2:  System information │
│ View: F3:[Playback] F4: Capture  F5: All             F6:  Select sound card  │
│ Item: Master                                         Esc: Exit               │
│                                                                              │
│                                     ┌──┐                                     │
│                                     │  │                                     │
│                                     │  │                                     │
│                          ┌───── Sound Card ──────┐                           │
│                          │-  (default)           │                           │
│                          │0  bcm2835 ALSA        │                           │
│                          │   enter device name...│                           │
│                          └───────────────────────┘                           │
│                                     │▒▒│                                     │
│                                     │▒▒│                                     │
│                                     │▒▒│                                     │
│                                     │▒▒│                                     │
│                                     ├──┤                                     │
│                                     │OO│                                     │
│                                     └──┘                                     │
│                                    72<>72                                    │
│                                  < Master >                                  │
└──────────────────────────────────────────────────────────────────────────────┘
```
2. Bluetooth: Bluetooth does not work and is not displayed in the control panel. There is an error message with the bluetooth service:
```
pi@bcm2711:/usr/local$ sudo service bluetooth status
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor pre>
     Active: active (running) since Tue 2020-05-05 10:35:34 EDT; 11h ago
       Docs: man:bluetoothd(8)
   Main PID: 608 (bluetoothd)
      Tasks: 1 (limit: 4044)
     Memory: 2.4M
     CGroup: /system.slice/bluetooth.service
             └─608 /usr/lib/bluetooth/bluetoothd

May 05 10:35:34 bcm2711 systemd[1]: Starting Bluetooth service...
May 05 10:35:34 bcm2711 bluetoothd[608]: Bluetooth daemon 5.52
May 05 10:35:34 bcm2711 systemd[1]: Started Bluetooth service.
May 05 10:35:34 bcm2711 bluetoothd[608]: Starting SDP server
May 05 10:35:35 bcm2711 bluetoothd[608]: Bluetooth management interface 1.14 in>
May 05 10:35:35 bcm2711 bluetoothd[608]: Sap driver initialization failed.
May 05 10:35:35 bcm2711 bluetoothd[608]: sap-server: Operation not permitted (1)
May 05 10:35:35 bcm2711 bluetoothd[608]: Failed to set privacy: Rejected (0x0b)
lines 1-18/18 (END)
```
Do you have any idea how to solve these issues?

pyavitz, 2020-05-06:
[SAP](https://www.gsmarena.com/glossary.php3?term=sap) stands for `SIM Access Profile` and can be disabled doing the following.
```
sudo nano /etc/systemd/system/bluetooth.target.wants/bluetooth.service
```
Change:
```
ExecStart=/usr/lib/bluetooth/bluetoothd
```
To:
```
ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap
```
Reload systemd:
```
sudo systemctl daemon-reload
```
Restart bluetooth:
```
sudo systemctl restart bluetooth
```
<hr>
As for the audio problem, i just tested my audio jack with some headphones and used moc to play an mp3, everything seems fine.<br><br>
Does Mate use pulseaudio? If so launch pavucontrol and check under Output Devices. It may be trying to push audio through the HDMI port(s).

alex9434, 2020-05-14:
Hi @pyavitz, thank you for your tipp! I got audio and bluetooth working. In case anyone stumbles into the same issue, I quickly document what I did:
A: Audio
1. disable pulseaudio
```
sudo cp /etc/pulse/client.conf /etc/pulse/client.confbackup
sudo nano /etc/pulse/client.conf
autospawn = no
```
2. Set headphone jack
`sudo amixer cset numid=3 1`

Now there is sound from the headphone jack. Unfortunately it is only mono:
```
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]

```
Do you have any idea how I can get to stereo output on the headphone jack?

B: Bluetooth:
`sudo apt-get install blueman`

closed 2020-05-15 by alex9434

pyavitz, 2020-06-05:
To force stereo do the following.
```
sudo nano /usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf
```
Comment out mono mode.
```
;[Mapping analog-mono]
;device-strings = hw:%f
;channel-map = mono
;paths-output = analog-output analog-output-lineout analog-output-speaker analog-output-headphones analog-output-headphones-2 analog-output-mono
;paths-input = analog-input-front-mic analog-input-rear-mic analog-input-internal-mic analog-input-dock-mic analog-input analog-input-mic analog-input-linein analog-input-aux analog-input-video analog-input-tvtuner analog-input-fm analog-input-mic-line analog-input-headset-mic
;priority = 7
```

alex9434, 2020-06-05:
Thank you! Will try.