pyavitz/rpi-img-builder #11: Avoid addressing of root device via /dev/mmcblk0p2

lategoodbye, 2020-05-17:
Hi,
i was using a prepared image for the RPi 4, but looking at the sources confirm that this issue still persists. Using a mainline kernel doesn't guarantee that the root device is always /dev/mmcblk0p2, because the RPi 4 has 2 different SDHCI controller and it's very likely that the root device is under /dev/mmcblk1p2 (depending on the probing order / kernel config).

PARTUUID could be a solution for this issue. Please look at https://github.com/gokrazy/gokrazy/issues/52 for possible inspiration.

pyavitz, 2020-05-17:
In my personal experience, I've only ever seen this happen when trying to use [mainline](https://www.kernel.org/). I believe it's the default in the dts file and is set that way because it's proper to do so. 

The foundation, must revert the change when they apply the patch set they use.

I'm not against using PARTUUID, I just haven't found a need to do so as of yet and seeing as this project isn't supporting [mainline](https://www.kernel.org/), I'm not seeing this as a problem.

Thank you for the above link, I'll give it a look see.

lategoodbye, 2020-05-18:
Currently i looking for an easy to use 64 bit rootfs environment for the Raspberry Pi 4 to test recent mainline work. So this project looks very promising.

pyavitz, 2020-05-18:
Well I may have a solution and I'm testing it right now. So far it seems to work, but I foresee trouble if loops are off and not clean.

```
echo 'BOOT_PARTUUID="' > boot1
blkid | sed -n '/loop0p1/s/.*UUID=\"\([^\"]*\)\".*/\1"/p' > boot2
paste -d '\0' boot1 boot2  > boot-id.txt
rm -f boot1 boot2

echo 'ROOT_PARTUUID="' > root1
blkid | sed -n '/loop1p2/s/.*UUID=\"\([^\"]*\)\".*/\1"/p' > root2
paste -d '\0' root1 root2  > root-id.txt
rm -f root1 root2

source boot-id.txt
source root-id.txt

### Cmdline
tee p1/cmdline.txt <<EOF
console=serial0,115200 console=tty1 root=PARTUUID=${BOOT_PARTUUID} rootfstype=ext4 elevator=deadline fsck.repair=yes logo.nologo rootwait
EOF

tee p2/etc/fstab <<EOF
PARTUUID=${BOOT_PARTUUID}	/boot		vfat    defaults 0 2
PARTUUID=${ROOT_PARTUUID}	/		ext4	defaults,noatime,nodiratime,commit=600,errors=remount-ro 0 1
tmpfs		/tmp		tmpfs	defaults,nosuid 0 0
EOF
```

Essentially it grabs and uses the PARTUUID's of the loop, which should in theory become the PARTUUID's of the image made. The image is about half way done cooking, so I'll let you know the results when done.

pyavitz, 2020-05-18:
I added a [mainline branch](https://github.com/pyavitz/rpi-img-builder/tree/mainline) to the hub and also a sample:  [`rpi-4-b-debian-buster`](http://www.mediafire.com/file/sa1vycew06ysb02/rpi-4-b-debian-buster-2020-05-18-v8.7z/file) `5.7.0-rc6-v8`

**Beware**:
If you are going to build an image of ur own, that "at this time" the loops need to be very clean. As in reboot and start the build.

Eventually I'll work out a fix this and I believe I already have one in mind.

~_Cheers!_

lategoodbye, 2020-05-18:
Thank you very much, i will try to test and report back

pyavitz, 2020-06-02:
I added a menu interface to the mainline branch.
Enjoy!

closed 2020-06-02 by pyavitz

pyavitz, 2020-07-13:
Mainline has now been integrated into the master branch.