pyavitz/rpi-img-builder #12: Prebuilt images

tichun, 2020-05-23:
There are two problems, images aren't hosted on github releases section but on Mediafire, that makes downloading them hard without GUI on the pi. 
Second problem is there are other files in the archive so it cannot be flashed without extracting it on disk, a big problem on sub-16gb cards.
Expected result is to have below command working:
`p7zip -d -c archive.7z | dd of=/dev/sdX Bs=4M conv=fsync status=progress`
Because currently it produces broken partition.

pyavitz, 2020-05-23:
I add a README file along with the *.img explaining the basics.

```
user	pi 
passwd	board

------------------

Execute: run-setup

run-setup will ask and perform the following:
		Locales, Timezone and Keyboard.

Change passwd(s): sudo passwd $USER

Webs:
Examples can be found in: /etc/network/interfaces
To setup wifi edit: /etc/wpa_supplicant/wpa_supplicant.conf
nano and mc are installed for easy editing and navigation.

Enjoy!
```
I could of course just add this to the hub, but I feel I would receive tons of unwanted issues if I did.

As for the release section? Point taken. In the future, I'll look to add future sample images there.

tichun, 2020-05-23:
`7z d archive.7z README` will do in described situation.

closed 2020-05-23 by tichun

pyavitz, 2020-05-23:
Sample images are now under release and the README has been removed.
Thanks for the input.