pyavitz/rpi-img-builder #14: Xfce4 Destop Image freeze after inactivity

rcla, 2020-06-17:
With test image 'rpi-4-b-xfce4-debian-buster-2020-06-06-v8.7z',
after a period of inactivity, it does not respond, it freezes.

I fixed it, installing the package 'xscreensaver' since in debian buster the package 'xfce4-screensaver' does not exist.

pyavitz, 2020-06-17:
I did notice that if `Power Manager > Security: Automatically lock the session: When screensaver is activated` is set, it does freeze and not come back. I'll add xscreensaver to the package list and rebuild the test image.

Thanx for the input and for testing.

pyavitz, 2020-06-19:
Bluetooth seems to be broken in the latest test builds. As soon as it gets resolved or I figure out what has changed, I'll upload some new images.

pyavitz, 2020-06-20:
Alright. Bluetooth has been sorted and I added a new image for Xfce4. 
You can find it under [Release](https://github.com/pyavitz/rpi-img-builder/releases/tag/5.4.44).

rcla, 2020-06-21:
Thanks for fixing it.

I tried the new version and it automatically applied the Numix theme, it looks very pretty.

In addition to running the audio in HDMI, without having to modify the file config.txt

closed 2020-06-21 by rcla