pyavitz/rpi-img-builder #15: Ubuntu image and Network Manager

rcla, 2020-06-17:
Testing the image 'rpi-4-b-ubuntu-20.04-2020-06-11-v8.7z',
I installed 'ubuntu-desktop', and have the 'Network manager' handle wireless networks,
modifying the following files:

/etc/network/interfaces:
```
### Interfaces
source /etc/network/interfaces.d/*
```

/etc/wpa_supplicant/wpa_supplicant.conf:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

#network={
#	ssid="mywifissid"
#	scan_ssid=1
#	key_mgmt=WPA-PSK
#	psk="wifipasskey"
#}
```

However, after those changes, the wired network does not work.

pyavitz, 2020-06-20:
I haven't loaded up the img, but I'm assuming you porbably need to disable networking - `sudo systemctl disable networking`

rcla, 2020-06-21:
> I haven't loaded up the img, but I'm assuming you probably need to disable networking - `sudo systemctl disable networking`

I tried, but it still doesn't work.

With ifconfig, eth0 does not appear.

pyavitz, 2020-06-22:
I fixed the problem by doing the following:
`sudo rm -f /usr/lib/NetworkManager/conf.d/10-globally-managed-devices.conf`

rcla, 2020-06-22:
I removed the file you suggest and now it works!

Thank you so much.

closed 2020-06-22 by rcla