pyavitz/rpi-img-builder #16: Make images USB Bootable for rPi 4

Pitriss, 2020-06-21:
I tried your test image for rPi4 and noticed that booting process have paths to devices hardcoded. It would make more sense to use UUIDs there.

pyavitz, 2020-06-21:
I've been working on this, but haven't come up with a stable way to implement it. That's why it's not currently in use, except under the mainline branch.

Creating sample images that use the UUID is simple enough and I could do so in the future, but adding the feature to the builder, with it not being stable, I feel will just create more headaches.

If you or anyone else know of a stable solution, feel free to create a Pull request or share ur code over at freenode. `#debianarm-port`

Thanx!

pyavitz, 2020-06-21:
Alright. I believe I have it sorted. I will update the builder in the coming days.
I added a new rpi4 sample img to [Release](https://github.com/pyavitz/rpi-img-builder/releases/tag/headless) for testing.

Thanx!

pyavitz, 2020-06-23:
This has since been resolved.
```
cmdline now uses PARTUUID where fstab uses UUID.
firstboot script now checks for the following: mmcblk0, mmcblk1 and sda
```
If you have further troubles, feel free to reopen the issue.

closed 2020-06-23 by pyavitz