pyavitz/rpi-img-builder #18: SHH in Devuan

markosc, 2020-08-30:
hello Just testing on a raspberry pi 3, but I have a question, I can't access by ssh, usually in x86 system I can activate it but for rpi sshd is complicated and it is by LAN

pyavitz, 2020-08-30:
SSH is enabled by default, so there isn't anything that needs to be activated. Which Raspberry Pi 3 model are you using? I'm currently SSHed into a 3B+ thats connected by LAN and all seems fine.

Have you tried SSHing in using hostname.local? `ssh username@bcm2710.local`

pyavitz, 2020-08-30:
After further investigation I found the problem was in the credentials script that is active on first boot. I made the following [adjustments](https://github.com/pyavitz/rpi-img-builder/commit/bda42d08e7e0b8a57cf3447e426ed24d64ce9884) and also created a new [image](https://github.com/pyavitz/rpi-img-builder/releases/download/images/rpi-3-devuan-beowulf-2020-08-30.tar.xz).

It also seems, that image had an issue regenerating new ssh keys during first boot. So it's very possible it was just a bad image?

Let me know if this resolves the problem. Thanx!

markosc, 2020-08-30:
I usually connect with putty to the raspberry pi 3b + with its assigned IP, in this case, when entering the sshd configuration and activating allow root login, the port and any ip, I cannot connect, it rejects the connection, I will download the last image generated and I'll see if I still have problems, thanks.

markosc, 2020-09-09:
thanks now I was able to connect through bcm2710.local and it looks good the system seems that devuan gives a plus to the raspberry pi 3

pyavitz, 2020-09-10:
Good. I'm happy it worked for you. If any other problems should arise related to this subject feel free to open the issue again.
Cheers!

closed 2020-09-10 by pyavitz