pyavitz/rpi-img-builder #19: Creating D

ptTrR, 2020-11-25:


pyavitz, 2020-11-25:
Sounds like a plan. Let me know how it goes.
_Thanx!_

ptTrR, 2020-11-25:
Ive flashed a bullseye 5.9.y image and i thought i have it done for the raspberry 3 and now wondering my nothing is booting XD.
My bad. 

Have to reset my pc soon, but i will do one more image.

over docker its much faster then via virtual box i reconigized. 
So yeah, dont laught. Thats my first compile/build trys. 

If you got a messenger for texting would be great, so i dont have to post everything in this thread.
And yeah, i think a docker image will be much more easy for must guys. 
its pulling debian10:slim and then installing all required dependencies and pulling into /build your github history. 

regards

pyavitz, 2020-11-25:
Do you use IRC?

You can talk with us/me directly by joining our channel on Freenode; #debianarm-port

pyavitz, 2020-11-29:
Hopefully this and the last commit solves the problems on the docker side of things. The intention is to eventually further optimize these builds, but its kinda tough as they are always in flux. I figure lets play it safe for now.

https://github.com/pyavitz/rpi-img-builder/commit/7283d2244637d7a5057d6432368f7cae7ee3eb58

closed 2020-12-01 by ptTrR