pyavitz/rpi-img-builder #21: armhf for the Raspberry Pi 4

Jorilx, 2021-01-07:
Would it be possible to add support for building an rpi4 image using the armhf architecture? It would allow to use debs directly from Raspbian (such as their rpi-optimized VLC, for example)

pyavitz, 2021-01-07:
It's possible, yes. As I've had several branches that could build one, but I ended up dropping support for it. Mostly because [Raspberry Pi OS](https://www.raspberrypi.org/software/) can already do this, I didn't feel like supporting the kernel anymore and it involved extra scripting.

The legacy branch can still build one, although I haven't tested it in a long time. It might be broke somewhere?

Is this something people are actually interested in? When considering the official OS is available and already very well supported.

Jorilx, 2021-01-07:
Well I'd like to build a Devuan armhf image, without systemd :D

pyavitz, 2021-01-07:
I see... I'll look into it and let you know. Give me some time and just leave the issue open.

Thanx!

Jorilx, 2021-01-07:
Great! Many thanks for your time!!

pyavitz, 2021-01-13:
I haven't had a chance to test everything, but Debian, Devuan and Ubuntu appear to be building fine. Take note that 'make helper' 'fetch' and 'docker' are currently not supported.

`git clone -b armhf https://github.com/pyavitz/rpi-img-builder.git`

Jorilx, 2021-01-13:
Wonderful, I'll try it in a few hours, many thanks again :)

Jorilx, 2021-01-13:
I can confirm that the image is working, thanks a lot!

pyavitz, 2021-01-13:
You are very welcome.

closed 2021-01-15 by pyavitz