pyavitz/rpi-img-builder #25: Ubuntu 21.04 Beta

ptr1337, 2021-02-16:
Since Ubuntu is providing theirself rpi 4 images, i think it should be possible include that into the script.

pyavitz, 2021-02-16:
I'm actually considering just dropping Ubuntu support. Like the foundation, they lean towards adding dependencies in which we don't need and it becomes difficult for us to support them. But unlike the foundation, the packages are in the repo they provide which then starts to conflict with the builds. Hence the reason I've started blacklisting things.

Frankly, it is just becoming a headache.

pyavitz, 2021-03-19:
Its been added to the master branch. This is still under development so don't expect miracles. https://github.com/pyavitz/rpi-img-builder/commit/81fc8cac62aa237d2a89790b34a26edc656cf6cd

closed 2021-03-19 by pyavitz