pyavitz/rpi-img-builder #28: rpi-img-builder now fails under Devuan Beowulf amd64

adam-burns, 2021-03-24:
Guessing this is related to 7a4f97fcf9265856db2f9b446ee90b81bc7b7b9c
# Install all dependencies:
sudo apt install build-essential bison bc git dialog patch \
dosfstools zip unzip qemu debootstrap qemu-user-static rsync \
kmod cpio flex libssl-dev libncurses5-dev parted fakeroot swig kpartx \
aria2 pv toilet figlet lsb-release xz-utils curl e2fsprogs btrfs-progs \
distro-info-data crossbuild-essential-arm64 crossbuild-essential-armel \
gcc-8-arm-linux-gnueabi gcc-9-arm-linux-gnueabi gcc-10-arm-linux-gnueabi \
gcc-8-aarch64-linux-gnu gcc-9-aarch64-linux-gnu gcc-10-aarch64-linux-gnu \
gcc-8 gcc-9 gcc-10 debian-archive-keyring debian-keyring make
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package gcc-9-aarch64-linux-gnu is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package gcc-9-arm-linux-gnueabi is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gcc-9-arm-linux-gnueabi' has no installation candidate
E: Unable to locate package gcc-10-arm-linux-gnueabi
E: Package 'gcc-9-aarch64-linux-gnu' has no installation candidate
E: Unable to locate package gcc-10-aarch64-linux-gnu
E: Unable to locate package gcc-9
E: Unable to locate package gcc-10
make: *** [Makefile:115: ccompile] Error 100

adam-burns, 2021-03-24:
On branch debian64:
`make ccompile
`
is successful.

make menu
   ```
 ┌────────────────────────────────────────────────────────────────────┐
    │ Results:                                                           │  
    │────────────────────────────────────────────────────────────────────│  
    │ ls: cannot access 'rpi-4*.img*': No such file or directory         │  
    │ ls: cannot access 'output/bcm2711/': No such file or directory     │  
    │ devuan-beowulf-rootfs-aarch64.tar.xz                               │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    │                                                                    │  
    ├────────────────────────────────────────────────────────────────────┤  
    │                             <  OK  >                               │  
    └────────────────────────────────────────────────────────────────────┘  

```

ptr1337, 2021-03-24:
Hey,

i checked the debian repo right now and it seems like they removed the packages from backports. Earlier they were in. Probably because they are releasing soon bullseye. 

The second solution is using docker, the only dependencies which you really have are "docker, docker-compose, qemu-user-static binfmt-support" on your host system. 
Then clone the repo: git clone https://github.com/ptr1337/arm-img-builder-docker
Change in the docker-compose.yml to your needed architecture (amd64 / arm64 ) 
Then a simple docker-compose up -d && docker-exec it arm-img-builder bash ; and you a in your host system. 
First update the repos with "make update". 
When joining the container youre automaticly in the container, then go into your builder which you want to use and run your commands. 
the benefit i that everything is mostly isolated from your system, so that your host is not getting ruined.

regards

ps: ill update the containers today that the repos are up2date again.

pyavitz, 2021-03-24:
Yes. It's related to the host change, kernel compilation and debootstrap.
As for the other error, I just ran a quick test (native) and all appears fine.

[image]

Let me build cross and see what it does. I'll get back to you.

Note: If your not trying to build a custom kernel and want to save yourself some time, you can do the following.
```sh
make config
make 2711
make rootfs
make image
```
This will produce a rpi4 *.img and bypass kernel compilation. `make commands` and `make helper` are your friend.

adam-burns, 2021-03-24:
Thanks for the feedback.

Using the make menu now for convenience, but yes, stepwise make commands will be wrapped soon into jenkins job.

For this use case of generating unofficial (but moving towards official) Devuan RPi images with rpi-build-image: the preferences would be:
- to re-build everything, including RPi Foundation kernel natively
- to build on non-Docker platform (this is a full time build host system after all)
- to have image build platform stability over time (preferably with Devuan as the build host)

pyavitz, 2021-03-24:
The debian64/v7 branch should be able to build just fine across hosts: Debian, Devuan and Ubuntu. These branches don't care what GCC version you are using and just use whats available.

So if you are using Beowulf for example, it is going compile the kernel against GCC 8.

As an aside, all kernels pulled down using `make helper` are compiled native. Target: Debian/Devuan stable.

ptr1337, 2021-03-24:
If you got enough sbc's i would personally go for a distcc cluster. 
The performance gain really noticeable and still natively compiled. 

Personally i used my pi4, hc4, mac m1 for distcc compiling and it was really fast. little bit slower then my 12 threads i7 8700k  desktop with cross. 

Also created then a cluster included my rig with the crosstool-ng cross compiler libary and the performance was on fantastic. (yeah, there was one cross "compiler" in the cluster")

pyavitz, 2021-03-24:
Well I ran three tests. One native in a docker container and two cross with Debian Buster and Devuan Beowulf. With Buster I just ran the `make all` command and in Beowulf I used the Menu Interface.

Looking at your second post it would seem that the kernel hit an error, which then cascaded and produced errors through out the rest of the build. The Menu Interface isn't setup to handle this and Dialog clears the screen, which is unfortunate.

I could be missing  a dependency in `ccompile` and `ncompile`. Is the following package installed? libelf-dev

adam-burns, 2021-03-29:
Again, thanks for the responses! 
```
root@devuan-streamer:~# apt list --installed | grep  libelf-dev

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

root@devuan-streamer:~# 
```
Your tip seems to be accurate.

I've installed libelf-dev, regenerated user file, and trying again. Will report back here with results.

Thanks again!

adam-burns, 2021-03-29:
> I could be missing a dependency in ccompile and ncompile. Is the following package installed? libelf-dev

Bingo! Fixes the issue.

closed 2021-03-29 by adam-burns