pyavitz/rpi-img-builder #30: [Testing] adding cacule scheduler

ptr1337, 2021-05-08:
Im using this scheduler on my desktop and have a much better experience in general responsive's and also under heavy load still smooth runing desktop. 
Ive tested to get the scheduler running on arm devices and it worked great so far, ive tested armv7h and aarch64. 
This needs more tested from users which are using the arm device as daily driver. 

Patches can be found here: 

https://github.com/hamadmarri/cacule-cpu-scheduler

if building for 32-bit arm devices, the 32-bit converter is also needed.

Following config options needs to be added to the default defconfig:

CONFIG_FAIR_GROUP_SCHED=y 
CONFIG_CACULE_SCHED=y

pyavitz, 2021-05-08:
The scripting is gonna need to be adjusted, but it should be good enough for testing purposes.
https://github.com/pyavitz/rpi-img-builder/tree/RIB-3.10

pyavitz, 2021-05-10:
Create a Mentions above or below Furthermore with the appropriate information concerning this patch set and move User patches to the Mentions section of the README.md. Add whatever further information people would need to know as far as **Tips and Tricks** concerning this patch set. This will place the responsibility firmly on the user and creator of said patch and other patches that may be of interest in the future. 

**Example:**
### Mentions
The CacULE CPU scheduler:
https://github.com/hamadmarri/cacule-cpu-scheduler

```sh
If building for 32-bit arm devices, the 32-bit converter is also needed.
Following config options needs to be added to the default defconfig:

CONFIG_FAIR_GROUP_SCHED=y
CONFIG_CACULE_SCHED=y
```
#### User patches

```sh
Patches "-p1" placed in patches/userpatches are applied during
compilation. This works for both Foundation and Mainline kernels.
```

pyavitz, 2021-06-05:
Referenced in the README

closed 2021-06-05 by pyavitz