pyavitz/rpi-img-builder #31: using clang/llvm with lto for kernel compiling

ptr1337, 2021-05-08:
Since Kernel 5.11 + lto can be used for compiling which could be a great performance improvement 
The compile time would be little bit longer. 

Also adding -mtune="x" for compile flags can improve the kernel

ptr1337, 2021-07-31:
still not implemented ? :x 

i using it since 5.13 on my rig and there is definitly a performance increase. 
However, also to retarded to use the new irc :D

pyavitz, 2021-08-01:
> still not implemented ? :x
> 
> i using it since 5.13 on my rig and there is definitly a performance increase.
> However, also to retarded to use the new irc :D

I just haven't gotten around to implementing it yet, as I've never personally used it and not 100% sure how to integrate it into the builder. I'll do some reading and asking around and see what I can do within the next week or so.

As for IRC, it's not rocket science or anything :)

ptr1337, 2021-08-01:
the new one is rocket science. 

i can help you if you want

pyavitz, 2021-08-01:
Well you know where to find me, if you wanna chat.

If you have a working example, you can also post it here or Projects, much like Tenkawa did with ccache.

pyavitz, 2021-08-03:
Ideas, thoughts and or changes?

```sh
git clone -b clang https://github.com/pyavitz/rpi-img-builder.git clang
```

ptr1337, 2021-08-03:
I try a  full compile. 

but mostly i would say changegs in the defconf for lto.

ptr1337, 2021-08-03:
-fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=4 -ffat-lto-objects -fuse-linker-plugin

maybe need thiis CC FLAAGS somehow

pyavitz, 2021-08-03:
> I try a full compile.
> 
> but mostly i would say changegs in the defconf for lto.

Trying to  figure out how to remove this warning:
```sh
dpkg-architecture: warning: specified GNU system type aarch64-linux-gnu does not match CC system type
aarch64-unknown-linux-gnu, try setting a correct CC environment variable
```
Happens on both cross and native, just a diff warning. Although the kernels appear to be fine after the build?

ptr1337, 2021-08-03:
BEcause Cross compiling.
Let me check, ill do native.

ptr1337, 2021-08-03:
https://github.com/torvalds/linux/blob/v5.13/arch/arm64/Kconfig

ptr1337, 2021-08-03:
Didnt found the LTO option in the config. 
its compiling now. ,  if i see lto compiling ill tell  you

pyavitz, 2021-08-03:
I made further adjustments to it and added missing depends. Cross is kinda sketchy using Focal, but running Native using Hirsute seems to work just fine.

ptr1337, 2021-08-03:
LLVM=1 LLVM_IAS=1
and IAS also please.

Also you need to adjust the defconfig for lto compiling.

closed 2021-08-03 by ptr1337