pyavitz/rpi-img-builder #32: wocker will be fixed tommorrow, sorry-

ptr1337, 2021-05-18:


pyavitz, 2021-05-19:
Besides the missing depends there seems to be a problem with debootstrap and x86_64. I haven't tested native yet? But so far when trying to debootstrap Debian/Devuan `anything above stable` it hits an error.

```sh
I: Extracting zlib1g...
W: Failure trying to run:  /sbin/ldconfig
W: See //debootstrap/debootstrap.log for details
make: *** [Makefile:230: rootfs] Error 1
```

I've been google searching solutions, but so far haven't found anything. This has been tested on Debian and Arch hosts. Same results.

pyavitz, 2021-05-20:
Found the problem: [commit](https://github.com/pyavitz/rpi-img-builder/commit/2ee0968e1ff72a129ea81801960ab2496fa66941)

After you update the image you can just close this out.

closed 2021-05-20 by ptr1337