pyavitz/rpi-img-builder #33: Unable to boot from the created images

cubit9-danielpeyer, 2021-08-02:
I could successfully create the images for Debian Buster and Devuan Beowulf. But both images fail to boot on my RPi4. I always end up with a message like this:

```
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Done.
Gave up waiting for the root file system device. Common problem:
  ...
ALERT! PARTUUID=45cba071-02 does not exist. Dropping to a shell.
```
Do I have to change something somewhere in a configuration file?

My OS is: Ubuntu 20.04.2 LTS x86_64

My steps to create the images:
1. make config (enter a username and password)
2. make menu
  - Selected RPi 4B
      - Make Kernel 
          - Make ROOTFS 
          - Make Image
3. Write the image with Balena Etcher to an SD card

cubit9-danielpeyer, 2021-08-02:
Found my problem. It works as expected on a fresh Ubuntu installation.

closed 2021-08-02 by cubit9-danielpeyer

pyavitz, 2021-08-02:
It looks like something was off with ur `/dev/loop`'s and the builder got confused and used the wrong PARTUUID when creating the img. This can happen if there are multiple loops in use or you had a build fail and didn't run `make cleanup` afterwards. In any case a reboot usually fixes the problem when all else fails.