pyavitz/rpi-img-builder #35: Make img error "ls cannot access rpi4.img"

Lukas0120, 2021-09-25:
First I want to thank you for this amazing project, really appreciate all the work you have put into this. 

I am trying to build a debian img for my rpi4 and compile the kernel with LTO and clang. Everything seems to be working perfectly but on the last stage, "make image" and at the end I get an error message. 

The Result page displays the following: "ls cannot access rpi-4*.img No such file or directory" 
There is an img folder with a debian image, can I just use this? Or how can I solve this issue? Thanks in advance

ptr1337, 2021-09-25:
I got a running LTO Image with Ubuntu. I'll going to try to create a image with LTO with Debian

pyavitz, 2021-09-25:
No that would mean something is failing during stage1 or stage2 of creation. This could be any number of things? Without knowing more I would suggest doing the following.

If at any point in time during image creation there is a fail, you need to make sure and run `make cleanup`. This removes the /dev/loops* and old bits of the image being created during that fail. After that, you can run `make image` again.

Furthermore its tough to diagnose a problem if using the menu interface, as Dialog clears the screen. So on next run do the following `nano userdata.txt` and set verbose to `verbose=1`, followed by running `make image` in the terminal.

**Note:**
If you wanna be really sure all the loops are clean, run `make cleanup` and reboot the host machine. The builder will sometimes get confused if there are to many /dev/loops*

Lukas0120, 2021-09-25:
Thanks for the detailed reply! I am running Ubuntu 21.04 on the host machine and cross compiling for rpi4 Debian. 

Also, I enabled FULL LTO in the kernel menu, and applied Cacule Full 5.14 patch. Could this be the reason? 

I will follow your advice and as soon as I am home I will run another build with the options you mentioned.

ptr1337, 2021-09-25:
No pr

> Thanks for the detailed reply! I am running Ubuntu 21.04 on the host machine and cross compiling for rpi4 Debian.
>
> Also, I enabled FULL LTO in the kernel menu, and applied Cacule Full 5.14 patch. [...]

Probally not the cause of cacule. i compile you one, but with -full. because RDB wont improve anything on the rpi4

pyavitz, 2021-09-25:
Well it could be anything? The menu interface is currently still in Beta and I don't ever use it myself. I use the base commands. Whats inside the `output/bcm2711` directory? Did the rootfs creation process succeed? If the kernel was created and the rootfs tarball is there, the only logical next step would be to look at what is failing in the staging process. It could be anything  from you caught a poor mirror or something during the `Compiling Headers ...` bit failed at the end of stage2.

My suggestion would be to check to make sure all the proper files are in place and from there see whats going on during the creation process by using the `make commands`.

Lukas0120, 2021-09-25:
Got it working now :) Changed to ubuntu hirsute and the image was created successfully. Going to see if it boots now! Thanks everyone for all the help, really appreciate that you're taking your time helping me. Still learning linux :)

ptr1337, 2021-09-25:
> Got it working now :) Changed to ubuntu hirsute and the image was created successfully. Going to see if it boots now! Thanks everyone for all the help, really appreciate that you're taking your time helping me. Still learning linux :)

Gonna delete my image :D

closed 2021-09-25 by ptr1337

Lukas0120, 2021-09-25:
Just a quick question once again x) Is it suppose to come with or without a desktop environment? Because it boots and I ran "run-setup"  And now I am in a terminal environment. Which is fine of course but is it suppose to be like that?

pyavitz, 2021-09-25:
Yes. As this is pretty much a one man show and it became to much of chore trying to work a DE for each different distro option. On top of that, I use almost all of my SBCs headless, so it seemed counter intuitive for me to continue to focus on it.