pyavitz/rpi-img-builder #38: make rpi3-image fails when nameserver 8.8.8.8 cannot be reached

ghpille, 2021-11-20:
I have my own nameserver, which is the only one that can be reached because of my firewall configuration.

I had to modify scripts/devuan-stage2 to use my nameserver in stead of 8.8.8.8 or 8.8.4.4.

pyavitz, 2021-11-20:
https://github.com/pyavitz/rpi-img-builder#bootrename_to_credentialstxt

It explains you can modify the written img and adjust said files before first boot.

pyavitz, 2021-11-20:
Hopefully this resolves said issue for you: https://github.com/pyavitz/rpi-img-builder/commit/58a0c551aa607c4eadebac88ba715132bb7ee367

After creating said userdata.txt file, review the file and adjust the variables as needed.

pyavitz, 2021-11-27:
I believe this was resolved. If it wasn't feel free to reopen the issue.

closed 2021-11-27 by pyavitz