pyavitz/rpi-img-builder #40: Devuan Beowulf (but not Chimaera) image building now reporting error

adam-burns, 2022-01-10:
I recently updated and merged the build scripts, then started to get image build failures

```
make image
make pri-image
make rpi2+3-image
make rpi3-image
```
all succeed for Chimaera, but fail for Beowulf with logs ending with
```
...
Creating config file /etc/ssh/sshd_config with new version
Creating SSH2 RSA key; this may take some time ...
2048 SHA256:2nKNWaxtnuvdcgxH+yjzqVcIKUqOF7FoDDArkSQ5RFU root@ubuntu-4gb-fsn1-1 (RSA)
Creating SSH2 ECDSA key; this may take some time ...
256 SHA256:mRRwT+jWcyvOA3piC6qB1mDUoybUHcmwW572q70H53E root@ubuntu-4gb-fsn1-1 (ECDSA)
Creating SSH2 ED25519 key; this may take some time ...
256 SHA256:PhEqJ2NwKdN2m/2efc3pq+JZ/EV59WHcJ2xBE50XBY8 root@ubuntu-4gb-fsn1-1 (ED25519)
invoke-rc.d: could not determine current runlevel
Processing triggers for libc-bin (2.28-10) ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Package firmware-linux is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package firmware-misc-nonfree is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package firmware-linux-nonfree
E: Package 'firmware-linux' has no installation candidate
E: Package 'firmware-misc-nonfree' has no installation candidate
E: Unable to locate package firmware-realtek
E: Unable to locate package firmware-ralink
```

Non-free seems to not exist for either distro?

```
# curl https://pkgmaster.devuan.org/merged/dists/chimaera/main/binary-arm64/Packages.gz | gunzip - | grep firmware-linux
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  8 10.3M    8  879k    0     0   496k      0  0:00:21  0:00:01  0:00:20  496kPackage: firmware-linux-free
Filename: pool/DEBIAN/main/f/firmware-free/firmware-linux-free_20200122-1_all.deb
 59 10.3M   59 6319k    0     0   818k      0  0:00:12  0:00:07  0:00:05  903kRecommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: dbus, dkms, file, firmware-ath9k-htc, firmware-linux-free, hw-probe, less, lsof, lvm2, mdadm, netutils, openssh-client, time
 69 10.3M   69 7375k    0     0   842k      0  0:00:12  0:00:08  0:00:04  928kDepends: progress-linux-base-system, cryptsetup-bin, keyutils, plymouth, firmware-linux-free, spectre-meltdown-checker, dmidecode, ethtool, gpm, hdparm, ntp, nvme-cli, parted, rng-tools | rng-tools5, smartmontools
Suggests: fonts-powerline-extra, firmware-linux-nonfree, firmware-misc-nonfree, iucode-tool, amd64-microcode, intel-microcode, firmware-ast, intel-isdct, supermicro-ipmicfg
 89 10.3M   89 9455k    0     0   881k      0  0:00:11  0:00:10  0:00:01 1021kSuggests: firmware-linux
Suggests: firmware-linux
100 10.3M  100 10.3M    0     0   909k      0  0:00:11  0:00:11 --:--:-- 1074k
```

```
# curl https://pkgmaster.devuan.org/merged/dists/beowulf/main/binary-arm64/Packages.gz | gunzip - | grep firmware-linux
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Package: firmware-linux-free
Filename: pool/DEBIAN/main/f/firmware-free/firmware-linux-free_3.4_all.deb
 54 10.0M   54 5615k    0     0  1273k      0  0:00:08  0:00:04  0:00:04 1273kRecommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: less, dkms, haveged, time, firmware-linux-free, iputils-ping, netutils, file, dbus, openssh-client, lsof
 90 10.0M   90 9295k    0     0  1257k      0  0:00:08  0:00:07  0:00:01 1187kSuggests: firmware-linux
Suggests: firmware-linux
100 10.0M  100 10.0M    0     0  1247k      0  0:00:08  0:00:08 --:--:-- 1191k
root@devuan-streamer:~# curl https://pkgmaster.devuan.org/merged/dists/beowulf/main/binary-arm64/Packages.gz | gunzip - | grep firmware-linux
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 10.0M    6  655k    0     0   617k      0  0:00:16  0:00:01  0:00:15  616kPackage: firmware-linux-free
Filename: pool/DEBIAN/main/f/firmware-free/firmware-linux-free_3.4_all.deb
 60 10.0M   60 6239k    0     0  1032k      0  0:00:09  0:00:06  0:00:03 1121kRecommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: firmware-linux-free, apparmor
Recommends: less, dkms, haveged, time, firmware-linux-free, iputils-ping, netutils, file, dbus, openssh-client, lsof
 94 10.0M   94 9711k    0     0  1069k      0  0:00:09  0:00:09 --:--:-- 1148kSuggests: firmware-linux
Suggests: firmware-linux
100 10.0M  100 10.0M    0     0  1069k      0  0:00:09  0:00:09 --:--:-- 1133k

```

adam-burns, 2022-01-10:
Ah, OK. Under Debian/Devuan they are in nonfree, not main.

On an existing Devuan Chimaera RPi3, changed `/etc/apt/sources.list` to include contrib & nonfree
```
deb http://deb.devuan.org/merged chimaera main contrib non-free
deb http://deb.devuan.org/merged chimaera-updates main contrib non-free
deb http://deb.devuan.org/merged chimaera-security main contrib non-free
```
and the packages are visible.

pyavitz, 2022-01-10:
The sources.list for Beowulf isn't the same as the one used in Chimaera.

The Chimarea sources.list I'm using:
```sh
deb http://deb.devuan.org/merged chimaera main contrib non-free
deb http://deb.devuan.org/merged chimaera-updates main contrib non-free
deb http://deb.devuan.org/devuan chimaera-proposed-updates main contrib non-free
deb http://deb.devuan.org/merged chimaera-security main contrib non-free
deb http://deb.devuan.org/merged chimaera-backports main contrib non-free
```

adam-burns, 2022-01-11:
> I'll add Bookworm and Daedalus.
> 
> Thanx.

Awesome. Thanks so much for your work in maintaining this.