pyavitz/rpi-img-builder #41: Cygwin support

AverseABFun-windows, 2022-02-22:
Could you add support for cygwin so us over on windows can enjoy this super cool thing you made?

arakeen, 2022-02-22:
If you install WSL2 you can use a Linux instance (Debian or Ubuntu type works best) inside it and it can create the images. You just need to treat the system as a cross compiler system since it is an x86 system and not arm based.

AverseABFun-windows, 2022-02-22:
I would but umm... I kind of broke wsl a while ago and I can't figure out how to fix wsl.

Sent from my SUPER OLD iPad that can't update ITS SO OLD

arakeen, 2022-02-23:
The problem is that you need to be emulating the actual os environment (kernel/libraries/binaries) from Linux and Cygwin does not do that. Cygwin is a layer that sits on Win32 to be an Linux API emulation layer. If you use IRC I can help you get WSL fixed however.

pyavitz, 2022-02-23:
Dual boot on a x86_64 or use an a SBC that is running a supported aarch64 distro. I don't have the Windows OS available to me and even if I did, I wouldn't go about making the builder compatible. Someone is more than welcome to do so, if they so wish.

AverseABFun-windows, 2022-02-24:
Okay so I'm available at 4:00 CST tomorrow and I will have to get IRC so 4:30 CST tomorrow?

Sent from my SUPER OLD iPad that can't update ITS SO OLD

AverseABFun-windows, 2022-02-24:
Okay I looked up what IRC is so I mean I will have to get a IRC client

Sent from my SUPER OLD iPad that can't update ITS SO OLD

arakeen, 2022-02-24:
You can use the web based client at https://web.libera.chat. Make sure you read the web pages about registering a nickname or you will not be able to join the support channel as it requires being logged into your account.

pyavitz, 2022-03-24:
As I don't plan on ever supporting Cygwin, this will be going into the rubbish. If someone feels like doing the work they are more than welcome too.

closed 2022-03-24 by pyavitz