pyavitz/rpi-img-builder #43: using make commit kernels in image build

djsftree, 2022-05-16:
Hi!

I have sucessfully buit a kernel using 

make commit board=bcm2711

The kernel and headers deb file appear in /output

How do I install them? Perhaps in uscripts /  run_function2 () ?

pyavitz, 2022-05-16:
I saw that a whiles back, but haven't found the time to look into it. If you have any ideas, you are more than welcome to do a PR.

djsftree, 2022-05-16:
Thanks!  Fixing those kind of things is a bit beyond me.

I'm still struggling with setting up the gpg secure keys this repo.  For now --allow-unauthenticated

https://github.com/djsftree/rpi-img-builder/blob/870e0c982185d4dfd79e726e1c5e6cb19b4f6f54/files/userscripts/uscripts#L43

pyavitz, 2022-05-16:
Example from [website](https://build.opensuse.org/project/show/home:bone11111:branches:science:EtherLab).

```sh
To use this repo please add a file etherlab.list in the directory
/etc/apt/sources.list.d

Using the Release key
1. Get the key
wget http://download.opensuse.org/repositories/home:/bone11111:/branches:/science:/EtherLab/Debian_11/Release.key
2. Add the key
sudo apt-key add Release.key
3. Add the package source url
deb [trusted=yes] http://download.opensuse.org/repositories/home:/bone11111:/branches:/science:/EtherLab/Debian_11/ ./
4. Update package list
sudo apt-get update

Content of the file in case you are not use the secure key:
deb [allow-insecure=yes] http://download.opensuse.org/repositories/home:/bone11111:/branches:/science:/EtherLab/Debian_11/ ./
```

pyavitz, 2022-05-16:
> In any case, thanks for your great imaging tool. Just booted!

You are very welcome.

closed 2022-05-17 by djsftree