pyavitz/rpi-img-builder #45: function after user creation

djsftree, 2022-05-25:
Hi!

https://github.com/pyavitz/rpi-img-builder/tree/master/files/userscripts

Could you add a function called after the user creation script?  I wish copy some items into the user home folder.

pyavitz, 2022-05-25:
sure. give me a bit.

pyavitz, 2022-05-25:
Couldn't we just move the function? https://github.com/pyavitz/rpi-img-builder/commit/adbd300d98231de2b453d7e81bafc94bcdf46d0a

djsftree, 2022-05-25:
perfect! thanks!

pyavitz, 2022-05-25:
Yeap. No prob.

djsftree, 2022-05-26:
What would be the best way of running pip as the installed user?

apt install -y python3-dev python3-pip python3-pytest procps libprocps-dev net-tools

as user

```
python3 -m pip install setuptools wheel
python3 -m pip install psutil==5.9.0 --no-binary :all: 
```

pyavitz, 2022-05-26:
Inside the chroot? Hmm. Not sure and it sounds risky. I use pip3 sparingly and when I do I use `sudo`. Which they say is not recommended, but I've never had a game changing moment.

pyavitz, 2022-05-26:
I shoe horned it in, but it should also work with the user script function. Do a `git pull` and add the following to ur script...

```sh
apt install -y python3-dev python3-pip python3-pytest procps libprocps-dev net-tools
sudo -u ${USERNAME} python3 -m pip install setuptools wheel
sudo -u ${USERNAME} python3 -m pip install psutil==5.9.0 --no-binary :all:
```
If you wanna be 100% you own it afterward: `chown -R ${USERNAME}:${USERNAME} /home/${USERNAME}`

```sh
result:
ls p2/home/${USERNAME}/.local/lib/python*/site*
psutil  psutil-5.9.0.egg-info
```

pyavitz, 2022-07-03:
closing because of inactivity

closed 2022-07-03 by pyavitz