pyavitz/rpi-img-builder #49: Kernel update does not work

Daviteusz, 2022-11-10:
Fetch command for updating kernel does not work for me.

I'm using this release:
rpi-4-debian-bullseye-5.15.45-v8-ext4-2022-06-18.img.xz 

I have latest userland, and eeprom.

```
$ sudo apt update #and upgrade
$ sudo reboot
$ ssh raspberrypi
$ fetch -u [done] 
$ fetch -2
Fetching rpi4-stable ...
[#ef13e7 47MiB/47MiB(99%) CN:1 DL:9.5MiB][done]                                                                                                                                                                                              

Something went wrong? Please report which Distribution and Release you are using.
```
Output from bash -x fetch -2
```
+ DOWNLOAD='aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false'
+ URL=https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/
+ RED='\e[0;31m'+ DOWNLOAD='aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false'
+ URL=https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/
+ RED='\e[0;31m'
+ GRN='\e[0;32m'
+ PNK='\e[0;35m'
+ TXT='\033[0m'
+ YLW='\e[1;33m'
+ SRED='\e[0;31m'
+ SGRN='\e[0;32m'
+ SYLW='\e[0;33m'
+ FIN='\e[0m'
++ curl -I https://github.com
++ grep 'HTTP/2 200'
 ]][ -n HTTP/2 200 
++ command -v wget
+ [[ -n /usr/bin/wget ]]
++ command -v aria2c
+ [[ -n /usr/bin/aria2c ]]
++ command -v pv
+ [[ -n /usr/bin/pv ]]
+ :
+ '[' 1 -eq 0 ']'
+ getopts 12brumUhf OPTION
+ case $OPTION in
+ ECHO=stable
+ getopts 12brumUhf OPTION
+ '[' stable = 5.15 ']'
+ '[' stable = stable ']'
+ echo ''

+ RELEASE=stable
++ grep -w bcm2708 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2709 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2710 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2711 /etc/opt/soc.txt
+ [[ -n BOARD="bcm2711" ]]
++ grep -w arm64 /etc/opt/soc.txt
+ [[ -n ARCH="arm64"
ARCH_EXT="arm64" ]]
+ BOARD=rpi4
+ rpi-linux
+ initrd_check
+ [[ -f /etc/initramfs/post-update.d/99-initrd ]]
+ :
+ prep
+ '[' -e /home/mateusz/.build ']'
+ sudo rm -fdr /home/mateusz/.build/rpi4-stable.sha256 /home/mateusz/.build/rpi4-stable.tar.xz
+ cd /home/mateusz/.build
++ wget -S --spider https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.tar.xz
++ grep 'HTTP/1.1 200 OK'
+ [[ -n   HTTP/1.1 200 OK ]]
+ echo -e '\033[0mFetching rpi4-stable\e[0m ...'
Fetching rpi4-stable ...
+ aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.sha256
[#894adb 0B/0B CN:1 DL:0B]+ aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.tar.xz          
[#1ccce4 47MiB/47MiB(98%) CN:1 DL:9.4MiB]+ echo_bdone                                                                                                                                                                                        
+ echo -e '\e[0;35m[\e[0m\e[0;32mdone\e[0m\e[0;35m]\e[0m'
[done]
+ echo ''

++ sha256sum --check rpi4-stable.sha256
++ grep OK
+ [[ -n '' ]]
+ echo -e '\033[0mSomething went wrong? Please report which Distribution and Release you are using\e[0m.'
Something went wrong? Please report which Distribution and Release you are using.
+ exit 0
+ GRN='\e[0;32m'
+ PNK='\e[0;35m'
+ TXT='\033[0m'
+ YLW='\e[1;33m'
+ SRED='\e[0;31m'
+ SGRN='\e[0;32m'
+ SYLW='\e[0;33m'
+ FIN='\e[0m'
++ curl -I https://github.com
++ grep 'HTTP/2 200'
 ]][ -n HTTP/2 200 
++ command -v wget
+ [[ -n /usr/bin/wget ]]
++ command -v aria2c
+ [[ -n /usr/bin/aria2c ]]
++ command -v pv
+ [[ -n /usr/bin/pv ]]
+ :
+ '[' 1 -eq 0 ']'
+ getopts 12brumUhf OPTION
+ case $OPTION in
+ ECHO=stable
+ getopts 12brumUhf OPTION
+ '[' stable = 5.15 ']'
+ '[' stable = stable ']'
+ echo ''

+ RELEASE=stable
++ grep -w bcm2708 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2709 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2710 /etc/opt/soc.txt
+ [[ -n '' ]]
++ grep -w bcm2711 /etc/opt/soc.txt
+ [[ -n BOARD="bcm2711" ]]
++ grep -w arm64 /etc/opt/soc.txt
+ [[ -n ARCH="arm64"
ARCH_EXT="arm64" ]]
+ BOARD=rpi4
+ rpi-linux
+ initrd_check
+ [[ -f /etc/initramfs/post-update.d/99-initrd ]]
+ :
+ prep
+ '[' -e /home/mateusz/.build ']'
+ sudo rm -fdr /home/mateusz/.build/rpi4-stable.sha256 /home/mateusz/.build/rpi4-stable.tar.xz
+ cd /home/mateusz/.build
++ wget -S --spider https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.tar.xz
++ grep 'HTTP/1.1 200 OK'
+ [[ -n   HTTP/1.1 200 OK ]]
+ echo -e '\033[0mFetching rpi4-stable\e[0m ...'
Fetching rpi4-stable ...
+ aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.sha256
[#894adb 0B/0B CN:1 DL:0B]+ aria2c -c --download-result=hide --console-log-level=error --disable-ipv6=true --summary-interval=0 --show-files=false https://github.com/pyavitz/rpi-linux/releases/download/gcc-11/rpi4-stable.tar.xz          
[#1ccce4 47MiB/47MiB(98%) CN:1 DL:9.4MiB]+ echo_bdone                                                                                                                                                                                        
+ echo -e '\e[0;35m[\e[0m\e[0;32mdone\e[0m\e[0;35m]\e[0m'
[done]
+ echo ''

++ sha256sum --check rpi4-stable.sha256
++ grep OK
+ [[ -n '' ]]
+ echo -e '\033[0mSomething went wrong? Please report which Distribution and Release you are using\e[0m.'
Something went wrong? Please report which Distribution and Release you are using.
+ exit 0
```

pyavitz, 2022-11-10:
Does the same thing happen when pulling LTS?

Daviteusz, 2022-11-10:
Yes, the result is the same

Daviteusz, 2022-11-10:
Okay, I found a solution 

`71 | if [[ `sha256sum --check $BOARD-$RELEASE.sha256 | grep "DOBRZE"` ]]; then`

I changed | grep "OK" to "DOBRZE"

The reason is a different message after checking the checksum when using a different system language. Now it works

pyavitz, 2022-11-10:
Hmm I see. This needs to be resolved to prevent errors in the future.

Daviteusz, 2022-11-10:
I updated the script and your fix solves the problem. Thanks

closed 2022-11-10 by Daviteusz

pyavitz, 2022-11-10:
Very good. Thank you for checking.