pyavitz/rpi-img-builder #54: ls: cannot access 'rpi-4-*.img*': No such file or directory

esakarya, 2023-06-10:
Hi i follow your guides.
when i try make image says : ls: cannot access 'rpi-4-*.img*': No such file or directory

Debug says; e: Package 'locales' has no installation candidate

what can i do? i am on ubunntu jelly fish

pyavitz, 2023-06-10:
I'm sorry to hear you are having issues, but that doesn't give me much to work with.

Turn on verbose in the userdata.txt file `VERBOSE=1` and use the make commands.
Example: `make image board=bcm2711`

As for locales, that is present in all three distros: [click](https://www.google.com/search?q=debian+package+locales&sxsrf=APwXEddKiPTzaMyUrtZJW6Yl2TD02ncOKg%3A1686429620052&ei=tN-EZJXaAoGkqtsPmLmJMA&ved=0ahUKEwiVhMTzx7n_AhUBkmoFHZhcAgYQ4dUDCBE&uact=5&oq=debian+package+locales&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQAzIGCAAQBxAeMgQIABAeMgYIABAIEB4yCAgAEIoFEIYDMggIABCKBRCGAzIICAAQigUQhgMyCAgAEIoFEIYDOgoIABBHENYEELADSgQIQRgAUNcQWKIWYK4ZaAFwAXgBgAHuAogBlg2SAQcwLjQuMS4ymAEAoAEBwAEByAEI&sclient=gws-wiz-serp)

Which Distro and Release are you trying to build an image for?

esakarya, 2023-06-11:
Hi than you for your quick answer.
I am try make image for debian 10 (buster)

and verbosed output is: Missing linux deb packages

pyavitz, 2023-06-11:
I don't believe the builder supports Debian Buster anymore, as it is EOL.

Did you install the builder [dependencies](https://github.com/pyavitz/rpi-img-builder#dependencies-for-debian-bullseye-and-ubuntu-jammy-jellyfish)?

esakarya, 2023-06-11:
U are right buster doesnt supprt

closed 2023-06-11 by esakarya