pyavitz/rpi-img-builder #57: Questions

ozzyrob, 2023-09-07:
First of all great bit of kit.

I've been working alongside rod with making images for bookworm for the RPi with Linuxcnc installed. Works like a charm.
One question, once we've built an image, is there a way we can jump into the (with qemu) rootfs of the image, get a shell, git pull the most recent Linuxcnc sources and do a build ?

Actually just starting the image via qemu is what probably the info I need.

All the Best
Rob

pyavitz, 2023-09-07:
This could be done, but I doubt it would be worth all the effort. Might be easier to look at the ROOTFS section of the build and create a linuxcnc tarball. Then when you go into stage2 of image creation you can run the update.

If you wanted to try mounting the img, you could use kpartx, add the mount points needed and go about it that way. But I think that would be more of a pain.

ozzyrob, 2023-09-07:
Cheers for your reply, in the mean time I found a cleaner solution to what I was suggesting.

Built a "pristine image"
Go through the whole losetup thing with the "pristine image", mount loop2p2 (as it was on my machine), copied the whole rootfs to a directory. Bind mounted dev & proc and the rest in the directory, copied over the qemu-static for aarch64, chroot into the directory & bang off I go just building Linuxcnc.

So far so good.

pyavitz, 2023-09-07:
Hey if that works, have at it :D

rodw-au, 2023-09-08:
I think there is a better way to achieve what Rob wants. ie to move the generated debs out of the chroot.
Becasue you can copy files to the output folder, we can test if the linuxcnc-dev folder we clone exist by looking for a known file
```
if [[ `ls p2/tmp/devel/linuxcnc-dev/scripts/rip-environment` ]]; then
    	git pull
else
        git clone https://github.com/LinuxCNC/linuxcnc.git linuxcnc-dev
fi
Then we can install the debs in the /tmp/devel/linuxcnc-dev folder in stage 2
```
See, pyavitz has been teaching me a bit about bash scripting on the side :)

The only problem with this great plan is  that  the linuxcnc-dev folder is blown away with the p2 folder at the start of make image. Maybe we need to move the folder out of the chroot and restore it  in the stage 1 script. I was hoping the new caching functionality would save our data. I wonder whats the best way to retain it?

PS. I just got the linuxcnc debs into the output folder! excited!

ozzyrob, 2023-09-09:
Happy with the way I've implemented things

closed 2023-09-09 by ozzyrob