pyavitz/rpi-img-builder #60: ls: cannot access 'rpi-4-*.img*': No such file or directory

RosiersRobin, 2023-12-19:
I haven't done anything special (except for some custom changes in the userpatches folder) and I keep getting (all of a sudden) the follwoing issue:

[image]

Any ideas what causes this? I tried:

- rebooting host
- make cleanup
- make purge
- make purge-all

Anything that can resolve my issue?

pyavitz, 2023-12-19:
Is it actually producing the img and Dialog just isn't seeing it or is the img creation process failing?

Dialog clears the screen so its hard to debug fails. If you run the commands manually it becomes easier to see whats happening.

NOTE: kernel and rootfs only needs to be run once. Unless of course you are building a new kernel revision or purged everything.
```
make kernel board=bcm2711
make rootfs board=bcm2711
make image board=bcm2711
```

You can set `VERBOSE="1"` in the userdata.txt file if you wanna see whats going on behind the scenes.

NOTE: `make purge-all` removes sources and debs, so unless ur sure you don't need those I wouldn't do that.

pyavitz, 2023-12-19:
I didn't have any issues with my build. What exactly are these kernel changes and did you make adjustments to the defconfig?

[image]

First img was built using `make all board=bcm2711` and the second img using the menu interface.

RosiersRobin, 2023-12-19:
Hmm, well I had an sytax error in a script I added and it did some funky things... Sorry for the panic I causes haha, the system works like a charm, but now I'm here, is there any option to change those skull icons on boot? 😅

pyavitz, 2023-12-19:
Disable or delete the patch in `patches/all/6.1.y/001-CUSTOM-drivers-video-logo_linux_clut224-ppm.patch` or change the loglevel in the cmdline.txt file from `=5` to `=1`.

To disable, rename it; `001-CUSTOM-drivers-video-logo_linux_clut224-ppm.disable`

What, you don't like skulls? :D

RosiersRobin, 2023-12-19:
They scared me alot first time booting 💀 💀 💀 

It's not THAT important it gets changed or removed tho, would be cool if there was an option to include it in the build, was just wondering. Thanks for the quick support! Keep it up!! 💀 ❤️

pyavitz, 2023-12-19:
> They scared me alot first time booting 💀 💀 💀
> 
> It's not THAT important it gets changed or removed tho, would be cool if there was an option to include it in the build, was just wondering. Thanks for the quick support! Keep it up!! 💀 ❤️

ur welcome

closed 2023-12-19 by RosiersRobin