pyavitz/rpi-img-builder #61: Commit not found

rodw-au, 2023-12-31:
I'm updating the linuxcnc builder to use kernel 6.1.69 for the pi4b and Pi5 but the commit is not found
I have rebased to your master today
I tried the commit after the merge and it is still not found.
```
ENABLE_COMMIT="1"
### Use this for the Pi4/4b and Pi 5 (6.1.69)
```
COMMIT= "342c7ee49e862edc30c893f141f55b9211b7a43b"

I get these messages in console
```
# Compiling kernel
userdata.txt: line 29: 342c7ee49e862edc30c893f141f55b9211b7a43b: command not found
Using commit: 
It appears the commit you selected is not available?
# Root Filesystem
userdata.txt: line 29: 342c7ee49e862edc30c893f141f55b9211b7a43b: command not found
```

It seems odd it reports a command not found.

rodw-au, 2023-12-31:
Hmm, changed  userdata.txt to not use a commit and pasted it in when asked and it seemed to have worked. odd

pyavitz, 2023-12-31:
Worked for me.

```
odroidn2plus: rpi-img-builder git:( master ?:12 ✗ ) $ make commit board=bcm2711
# Compiling kernel

Using commit: 342c7ee49e862edc30c893f141f55b9211b7a43b

Downloading linux-rpi-6.1.y ...
[#af8ef2 211MiB/0B CN:1 DL:7.4MiB] [done]                                        

Extracting archive ...
 212MiB 0:00:36 [5.76MiB/s] [==================================>] 100%            

Firmware [downloading] [checking] [done]
Patches [applying] [done] log: output/logs/bcm2711-patch.log

Making bcm2711_defconfig.
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/conf.o
  LEX     scripts/kconfig/lexer.lex.c
  HOSTCC  scripts/kconfig/expr.o
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#

```

rodw-au, 2023-12-31:
Odd. Must be spurious white space or something. I'll try again when it stops building what I want :)

OFFTOPIC: What is the chances of building a PREEMPT_RT for an Odroid N2+ (I got one here)

pyavitz, 2023-12-31:
I don't see why it can't be done.

rodw-au, 2023-12-31:
I'll give it a go after I build these Pi images for a later linuxcnc version . Geez, I don't even have a Pi!
I had no luck with the Odroid on the Armbian builder so been meaning to try yours...

closed 2024-03-13 by pyavitz