pyavitz/rpi-img-builder #62: Can't build compressed image

rodw-au, 2023-12-31:
Six weeks ago before you adopted rsync for compression, I built compressed  images for the Raspberry Pi5 and Pi4b
Image sizes were 
pi4b 982.5 Mb
pi5  988.4 Mb

Regardless of the image size set in custom.txt, I get a no space left error
I can build an uncompressed image and its 32.8 GB which is way larger than we experienced in the past. (about 4 gb)

```
# Image Size: 3584MB 4096MB
IMGSIZE="98304MB"
```
Error is
```
Shrinking [4723MB]rsync: [receiver] write failed on "/home/rod/devel/rpi-img-builder-2.9.2/p2/usr/share/icons/Adwaita/cursors/watch": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(380) [receiver=3.2.7]
rsync: [sender] write error: Broken pipe (32)
make: *** [Makefile:132: image] Error 11

```
I can't find the file referred to in the error. Its not something specifically installed by our program,
My build scripts have not changed other than to merge in your latest master branch and alter the userdata.txt and custom.txt

It appears somehow, your new build script is including additional folders or something given the small size of our prevoius images.

If you can investigate and advise, it would be great!

pyavitz, 2023-12-31:
https://github.com/pyavitz/rpi-img-builder/commit/1458dc8b28ba141dd11afa5fd00a06025c90b508

rodw-au, 2024-01-01:
[image]
Translated:
The input file is not a valid disk image.
The file size 4723000000 bytes is not a multiple of 512 bytes

pyavitz, 2024-01-03:
This should be resolved.
https://github.com/pyavitz/rpi-img-builder/commit/eae1e81a3290e51c5104ff491f2a7ce2d6b31783
https://github.com/pyavitz/rpi-img-builder/commit/eb0e1108eb23956097f2f4a468abf9a15bbad336

closed 2024-03-13 by pyavitz