pyavitz/rpi-img-builder #63: chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error

smoe, 2024-01-08:
Hello, I am cross-building from WSL running Debian and it looks surprisingly fine. I now ran into the following error which I presume not to be triggered by me starting non-natively. Does this trigger any immediate thoughts? Would you want me to test anything?
Many thanks!
Steffen
```
I: Extracting tzdata...
I: Extracting usr-is-merged...
I: Extracting util-linux...
I: Extracting util-linux-extra...
I: Extracting zlib1g...
chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error
make[1]: *** [Makefile:149: all] Error 126
make[1]: Leaving directory '/home/moeller/rpi-img-builder'
make: *** [Makefile:212: menu] Error 2
```

smoe, 2024-01-08:
Ah - I started out with Debian trixie, should distributions be hard-coded anywhere ... started digging :)

pyavitz, 2024-01-08:
I've seen that error before `chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error` but not in a really long time. If i recall it usually happened in docker?

Whats the host distro ur running? You say you started with trixie. Do you mean thats the host ur using or the rootfs you are trying to create?

I don't use Windows or VM's, so I don't have any tips or tricks on how to make that work. I either build native on aarch64 or cross on x86_64.

pyavitz, 2024-01-08:
> The system is somehow not informed about the auto-invocation of qemu.
> 
> ```
> $ sudo update-binfmts --display
> python3.11 (enabled):
>      package = python3.11
>         type = magic
>       offset = 0
>        magic = \xa7\x0d\x0d\x0a
>         mask =
>  interpreter = /usr/bin/python3.11
>     detector =
> python3.12 (enabled):
>      package = python3.12
>         type = magic
>       offset = 0
>        magic = \xcb\x0d\x0d\x0a
>         mask =
>  interpreter = /usr/bin/python3.12
>     detector =
> ```
> 
> and somehow this seems related to [microsoft/WSL#2103](https://github.com/microsoft/WSL/issues/2103) - tried around but failed. The location of format descriptions also seems to have changed:
> 
> ```
> $ dpkg -L qemu-user-static|grep binfmt | xargs -r dirname | sort -u
> /usr/lib
> /usr/lib/binfmt.d
> /usr/libexec
> /usr/libexec/qemu-binfmt
> ```
> 
> Nothing urgent on my side - I happily use Rod's binaries.

Nothing urgent on my end either. All of which you are trying to do, is not supported by the builder.

closed 2024-01-30 by pyavitz