pyavitz/rpi-img-builder #65: Unabel to see /dev/ttyAMA0

RosiersRobin, 2024-01-12:
Hi,

I compiled the image and it worked out of the box, no biggy, however it seems I cannot see the /dev/ttyAMA0 device. It's a RPI hat plugged in (qibixx). Anyone an idea? I normally can access it via minicom and run some serial command. Now, I cannot. /dev/TTY0 doesn't do much either.

Any ideas here?

pyavitz, 2024-01-13:
Did you get this resolved?

RosiersRobin, 2024-01-15:
I got it resolved by disabling the bluetooth in the configs;

[image]

However, the serial port is acting up weird. I tried the same setup with the same hardware, but with Ubuntu server 22.04 and it didn't act up.

this is what hapens:
[image]

RosiersRobin, 2024-01-17:
i'm going to try and compile with the lower kernel version and see if the issues still persist. If they don't, then there must be something with the kernel going on I think..?

RosiersRobin, 2024-01-18:
Tested with an older kernel, seems like it's still the same "laggy" serial. I'll try different hardware.

RosiersRobin, 2024-01-18:
Exactly the same with different hardware. On the RPI3 modebl B+ it works like a charm (with raspbian) but on the 4 it does not.

pyavitz, 2024-01-18:
I have no clue what the solution would be. It could be something simple, but without the HW myself I can't debug.

RosiersRobin, 2024-01-18:
Compiling using debian instead of ubuntu as distro resolved the issue.

closed 2024-01-18 by RosiersRobin