pyavitz/rpi-img-builder #66: Multi board image

jayofelony, 2024-02-13:
Would it be possible to build an image for multiple boards at the same time?
02w, 3, 4 and 5 all at once arm64 only with kernel 6.6.y.

pyavitz, 2024-02-13:
Technically the RPI4 `aarch64` `bcm2711` img should be able to boot those.

jayofelony, 2024-02-14:
Can you make it add the bcm2712-rpi-5.dtb to it as well when you build an image for pi 4?

pyavitz, 2024-02-14:
> bcm2712-rpi-5.dtb

Is it not there? It should be.

jayofelony, 2024-02-14:
When i build a pi 4 image there is no pi5 dtb there

pyavitz, 2024-02-14:
That's because ur using 6.2.y and the DTS wasn't introduced. That kernel is EOL and the PI5 DTS wasn't added until after the fact.

It was added to 6.1.y because they are still using that kernel as an LTS. Eventually they will move on to 6.6 LTS and drop support for 6.1.y.

closed 2024-03-13 by pyavitz